stevenbennett/Open-Task-Light #1: LED Housing alternatives?

jay-jey, 2022-12-02:
Hello, it seems that the housing of the LED is not longer availible, do you know any other housing that is working with the Lens and LED? There are so many different options, it is one of the smallest part but I think very important. Hope you cloud help me out with this. 
BTW, very cool project.

stevenbennett, 2022-12-02:
@jay-jey Hold tight for a bit because I've recently changed the LED and corresponding LED housing (due to issues like this where components abruptly become unavailable). Once I have the new LED tested I will update the BOM and this issue :)

stevenbennett, 2023-01-05:
@jay-jey Just updated the parts list with the new LED housing, and it's in stock woo!

stevenbennett, 2023-01-05:
It's this one btw: https://www.mouser.com/ProductDetail/Ledil/F15256_HEKLA-SOCKET-D?qs=o2C7DKDMuStSi5P8paemqw%3D%3D

closed 2023-01-05 by stevenbennett