stevenbennett/Open-Task-Light #10: linear power rail

fabianmuehlberger, 2023-01-24:
Hey Steven, 

I watched your videos on youtube and the video about the drag chain made me think about other solutions for supplying power to the head of the lamp. 

So what if you could include the conductor in the aluminum profile? One way would be to coat the aluminum profile with PCB lacquer (eq. kontakt chemie plastik 70. ) and glue a flat aluminum conductor inside the slot. This would make it relatively safe to touch, assuming you only use 12V this isn't an issue. Then use copper brushes on the sliding part of your assembly to transfer the electricity (For testing you could use a solder wick, it comes in different sizes and is cheap, or look for replacement parts of the Carrera Toys). Technically you could use the aluminum profile as the earth, but having 1 conductor on each side where your wheels sit is an option. 
You would need to build some tensioning spring to keep the brush engaged.

With the PCB lacquer being transparent and if you use an aluminum conductor, you won't have any visible cables, the overall part count is reduced and properly made I assume it can be very reliable over time with low maintenance besides replacing or adjusting the brushes. 

I have some aluminum profiles in my workshop and can do some testing if you are interested.

Best 
Fabian

fabianmuehlberger, 2023-01-24:
Just noticed the discussion feature. Therefore closed the issue

closed 2023-01-24 by fabianmuehlberger

stevenbennett, 2023-01-30:
@fabianmuehlberger All good, thanks for sharing 👍🏻