stevenbennett/Open-Task-Light #12: Filled Shopping Cart Links and Eccentric Spacer dimensions

momonunu, 2023-02-01:
This would be a link to the full openbuildstore cart for the diyson express (including the  end caps): https://saveyourcart.io/cart-url/qZya83ZUsDFTxaBH/db371481-3787-42a3-bb20-5ef49feb6bf3
I was not sure if the eccentric spacer is the 6mm variant or the 1/4“ version. In the cart above it is the 6mm version.

I am not sure however if the referral link gets saved in the saved cart.

tobias-holmes, 2023-02-01:
Thanks for sharing the saved cart. I noticed it was missing the 40 mm spacer to connect to the desk mount. Here is an updated cart: https://saveyourcart.io/cart-url/qZya83ZUsDFTxaBH/88853f4a-3ad2-4483-a033-878baa2b9b02.

I'm also assuming the 6 mm version of the eccentric spacer based on the step files. Please correct me if I'm wrong.

stevenbennett, 2023-02-01:
@momonunu @tobias-holmes Thanks for this! Yeah not sure if this will work with my referral link but this is just too convenient to pass up.

Best of both worlds; could this be set up using my site-wide referral link? https://openbuildspartstore.com/?ref=_2E7m4SXRIy8t

Not a huge deal if these things are incompatible, but I was hoping the referral commissions could help offset the project costs.

stevenbennett, 2023-02-01:
@momonunu @tobias-holmes And yes it's the 6mm eccentric spacer

momonunu, 2023-02-01:
Probably best to use the cart link first and then the sitewide referral in order to support you.

closed 2023-02-08 by momonunu