stevenbennett/Open-Task-Light #22: Consider securing the Base part with a through-bolt

dflemstr, 2023-02-13:
The Base part currently consists of a short bolt to attach it to an aluminum extrusion, and then the 40mm spacer to attach it to the Desk Mount.  This makes it quite brittle.

Gonna try to visualize using ASCII art, the drawing is not to scale :)

```
                                                      Bolt

                            ┌──────────────────────────┐ ┌───────────────────────────────┐
                            │                          │ │                               │
┌───────────────────────────┼───────────────┐   ┌───┐  │ │                               │
│                           │               │   │   ├──┼─┼─────┐                         │
│                           │               │   │   │  │ │     │                         │
│                           │               │   │   ├──┼─┼─────┘                         │
└───────────────────────────┼───────────────┘   └───┘  │ │                               │
                            │                          │ │                               │
40mm spacer                 └──────────────────────────┘ └───────────────────────────────┘
                                   3D printed base               Aluminum extrusion
```

An alternative is to use a long bolt that goes through both the 40mm spacer and through the Base 3D printed part and all the way into the aluminum extrusion.

```
Bolt                   ┌──────────────────────────┐   ┌─────────────────────────┐
                       │                          │   │                         │
┌───┐  ┌───────────────┼────────────────────────┐ │   │                         │
│   ├──┼───────────────┼────────────────────────┼─┼───┼────────┐                │
│   │  │               │                        │ │   │        │                │
│   ├──┼───────────────┼────────────────────────┼─┼───┼────────┘                │
└───┘  └───────────────┼────────────────────────┘ │   │                         │
                       │                          │   │                         │
                       └──────────────────────────┘   └─────────────────────────┘
        40mm spacer           3D printed base                 Aluminum extrusion
```

I've managed to do this with a 55mm or 60mm bolt quite easily with some minor tweaks to the 3D model using slicer hacks, so I think it should be quite straight-forward to update the original model as well.

stevenbennett, 2023-02-14:
@dflemstr Cool idea, this was one of my initial concepts actually (though I never mentioned it in any build logs). Since you have this implemented already maybe you can answer a question for me; does the screw ever unseat itself when turning the lamp counter-clockwise? Early on I worried that the friction on the screw head and a counter-clockwise turn might be enough to unscrew it, but perhaps that weight of the lamp prevents this. Any findings here?

Also thanks for the art, I like these :)

stevenbennett, 2023-02-14:
Oh sorry just reread and realized you may not have actually implemented this yourself. Maybe somebody can give it a try