stevenbennett/Open-Task-Light #23: Consider supporting 6mm belts

dflemstr, 2023-02-13:
It seems to be way easier to find 6mm belts compared to 5mm belts, and it also seems to fit fine into all the aluminum extrusions I had laying around.

legopc, 2023-02-13:
Yeah I agree. I ordered 5mm belts just to be sure but I might also order some 6mm to compare. 6mm does fit, is easier to find locally so would be a good replacement for the partslist.

stevenbennett, 2023-02-14:
@dflemstr @legopc I'll keep it in mind. The nice thing about the 5mm belts is that as far as I know you can often find them where v-rollers and rails are sold. Though I appreciate that outside of that scenario that that 6mm belts are easier to find. Thanks for flagging that.

The 6mm belt will fit on paper, but there's a chance it will rub agains the aluminum rails. If either of you want to test out a 6mm belt let me know and I can send you the Fusion 360 files for the relevant parts and you can give it a go.