stevenbennett/Open-Task-Light #41: Safer counterweight material

ekostjuk, 2023-03-16:
Why not use solid square tungsten/carbide bar (same or more densite than lead, 100% packed/efficient) + top-up with steel/copper balls for balance?

jfuchs, 2023-04-16:
I also had this concern. My current plan is to use fender washers like [these](https://www.amazon.com/Hillman-290012-Fender-Washer-100-Pack/dp/B000BD5GFU/ref=sr_1_3?crid=2LIJVYAL7NK6F&keywords=hillman+fender+washer+1+1%2F4&qid=1681675920&s=hi&sprefix=hillman+fender+washer+1+1%2F4%2Ctools%2C129&sr=1-3) because they're easy to adjust and pack densely. I'll have to design a new enclosure for the counterweight, which will need to be thicker, but not as long.

jfuchs, 2023-04-17:
> I have kids running around the house, so anything lead is a no-no.

Yeah, similar concerns for me.

The nice thing about fender washers specifically is that their bore hole diameter is a small percentage of their outside diameter, so if you have them stacked, you're pretty close to packing the space as much as possible — at least if you're okay with the counterweight shape being dictated by the washer. I'm considering something like this:

[image]

But on the other hand, I recognize that this breaks with the rest of the design language. Suggestions welcome.

legopc, 2023-04-17:
Maybe we should take a page from monitor stands a take a look at springs as counterweight?

jfuchs, 2023-04-17:
> Maybe we should take a page from monitor stands a take a look at springs as counterweight?

As I understand it, a spring's force is going to vary with its displacement — but in this case i think you'd want it to be constant across different displacements. Maybe there are kinds of springs I don't know about though!

ekostjuk, 2023-04-18:
Long enough pre-loaded spring/elastic might be stable enough vs minimal friction across limited range of movement. 
I'll keep looking for tungsten stock, as front plate is also done from polished bronzefill, which adds.

JeSuisAnanas, 2024-10-04:
How about filling the 3d printed part with sand or a sand epoxy mixture before printing the top layers?

stevenbennett, 2025-11-02:
Did some testing and I found a safe and affordable material using this [steel abrasive blasting media](https://amzn.to/4nxTcas). 

I plan to have two variants of the counterweight (one for lead and one for steel) as the steel option requires a larger volume. Here in the states you can get 5lb of this stuff shipped for ~$23 which is slightly more than required for the freestanding base build. The counterweight needs to weigh ~1lb 1.5oz (497g) for the final build (assuming a 500mm horizontal rail and a 3D printed carriage).

The final counterweight design will be uploaded in the coming months once the project is finished, but my overall external dimensions using this material is 205x40x20mm. For the leaded version just the depth of the counterweight will change so it will be more like 205x30x20mm. The link is an affiliate link but use whatever works for you!